forth32/balong-usbdload #2: B310

hfbus, 2016-07-13:
Hello, i search the file usblsafe-b310.bin, Добавлен загрузчик B310. Thank you

forth32, 2016-07-13:
You can send original usbloader.bin to me, i will attempt to path it.

-- 
Alexander Stepanov

hfbus, 2016-07-13:
i don't have original usbloader.bin, how can i help you ?

hfbus, 2016-07-14:
maybe it's the same bootloader for b310 and b315.

wsw243, 2016-08-02:
http://www.mediafire.com/download/bdqs3hkez2awkh3/b310_usbloader.rar original b310 bootloader patch it

wsw243, 2016-08-02:
Also patch e8231 bootloader http://www.mediafire.com/download/08f456dzj7d5s4m/E8231_wingle_usbloader.rar

hfbus, 2016-08-05:
@forth32 thank you

alirazam, 2016-08-15:
forth32 plz http://www.mediafire.com/download/bdqs3hkez2awkh3/b310_usbloader.rar original b310 bootloader patch it

mohammadmsh, 2016-10-11:
forth32 plz I need to -usblsafe-b310.bin- thank you

forth32, 2016-10-11:
added usbloader-b310.bin

-- 
Alexander Stepanov

mohammadmsh, 2016-10-11:
forth32 thank you But does not work -b310s-22 - can you help me

forth32, 2016-10-11:
What it means - "does not work"?
b310 does not have usb ports. How you can load usbloader without usb?

-- 
Alexander Stepanov

mohammadmsh, 2016-10-11:
The router does not work I want to download a bot for USB, because I've welded via Link Router

praveen02008, 2016-12-08:
Also patch e8231 bootloader

bezza06, 2016-12-10:
forth32
there is a port in B310s -xxx but b310loader does not work

udokaezepue, 2016-12-25:
Download all the files from here http://www.xseduct.info/2016/12/download-b310susbloader-balong-usb.html?m=1

niccowahyudi, 2017-02-09:
someone  have usb loader for e5372 ...! share with me pls..!

bezza06, 2017-02-09:
Hello i'think you can use usbloader.bin universal you find it in the zip
file you download

niccowahyudi, 2017-02-09:
there is no usbloader.bin in the zip..! can you give me the link..? thank you..!

bezza06, 2017-02-12:
this is the url

bezza06, 2017-02-12:
https://my.pcloud.com/publink/show?code=XZb6vlZifQGLLCGSTFJDt43cuJM6Lgn85Uy

niccowahyudi, 2017-02-12:
thank you bezza but it doesn't work ..! it making balong usb downloader not responding..!

bezza06, 2017-02-13:
hey give me  your  fb ,that means there is errors in the port numbre or your huawei is not in emergency boot

balwinder27, 2017-03-04:
https://drive.google.com/file/d/0B3pEUGa_GFBrS01XLU4zM2I1RHc/view

e8372h-153 zong free solution for everyone

closed 2017-09-24 by forth32