forth32/balong-usbdload #4: how to backup and restore nv data for  hiscilicon based modem

alirazam, 2016-08-11:
sir i have litle isshu 
ishu is that how to backup and restore nv data for hiscilicon based modem like e8231 e5330
i use usbloader-e8231.bin but after change firmware all nv data wipe even signal not com(no service)
kindely solve my problem .
or have another solution ?

forth32, 2016-08-11:
>Четверг, 11 августа 2016, 9:46 +03:00 от alirazam <notifications@github.com>:
>
>sir i have litle isshu 
>ishu is that how to backup and restore nv data for hiscilicon based modem like e8231 e5330
>i use usbloader-e8231.bin but after change firmware all nv data wipe even signal not com(no service)
>kindely solve my problem .
>or have another solution ?
Durig boot, usbloader erases all flash, including nvram partitions (nvimage, nvdefault, nvdload, nvbacklte). To avoid this problem, we must use patched version of usbloader (usblsafe). I created some safe usbloaders, but  i  never did safe  bootloader for E8231 . You can patch this biitloader by youself.

You can also backup nvram before trying to boot usbloader. Save file /mnvm2:0/nv.bin, and restor it after firmware flashing.

-- 
Alexander Stepanov

alirazam, 2016-08-12:
kindely sir guide me fully how to backup and restore and which tools i need.and whats method?

alirazam, 2016-08-14:
plz help me forth32 i wipe my nv about 10 devices.

abassi94, 2016-12-20:
me  to i need nvram  backup pls

MrHackmed, 2017-01-18:
@forth32 Please can you explain how to backup n restore in details?, 

I saw the command you wrote up there but I don't understand how to apply it. I have tried flashing the device severally with balong, and it's not coming up, just stuck on White power light and the notification showing at the taskbar is;

_The Last USB device you connected to this computer malfunctioned, and Windows does not recognize it._

I would be glad if you can do this soon. Thanks

closed 2017-09-24 by forth32