forth32/balong-usbdload #6: (Запрос) добавить HUAWEI b310-927 usblsalf.bin погрузчик

alirazam, 2016-10-08:
сэр Я прошу, пожалуйста, добавьте Huwei b310-927 usbslaf.bin грузчиком в проекте .в 20.313.xx ФВ не поддерживает пайку так USB шоу, как USB не распознает .only решение тест point.only вы можете патч загрузчик
я прошу пожалуйста, пожалуйста, загрузите usblsalf-b310.bin погрузчик в нашем проекте

forth32, 2016-10-08:
You can use non-ptached usbloader.bin for this router (get it form firmware packages). This progam now can patch it on the fly, during usbboot process.
But, use only last version of balong-usbdload. Old versions do not contain autopach functions.

alirazam, 2016-10-08:
уважаемый сэр плз помогите мне я e5330bs-2 я использовать не pached usbloader.bin
но после прошивки все данные нановольта удалить имя WiFi, изменение IMEI (NULL), а не обслуживать шоу (вставить SIM),
Далее я хочу восстановить NV Data плз я буду к вам благодарностью
скажите мне простой метод, как резервное копирование и восстановление данных NV На Hiscilicon на базе процессора
делает qpst работу инструмента для этого?

forth32, 2016-10-09:
Try to use fastboot interface of usbloader:
1. load usbloader in fastboot mode (balong-usbdload -f usbloader.bin)
2. Restore nvdefault partition from backup image (fastboot flash nvdefault nvdefault.img)
3. Erase nimg partition (fastboot erase nvimg)
4. Reboot router

After this, kernel will restore workin nvram (nv.bin) form nvdefault.

alirazam, 2016-10-13:
i run normal loader from this method
(shift+rignt click) open cmd here .
next type balong_usbdload -p usbloader.bin
next 2 port exist 
1>>pc ui interface
2>> application interface
so my question is that how to run usbloader in fastboot mode?

chipoza, 2016-11-07:
i backup nvdefault.img from my one device and how to restart in another device

alilihicham, 2016-12-20:
How do I get nvdefault.img

faridmay, 2016-12-21:
yaw mayhalbouch hna

alilihicham, 2016-12-21:
hhhhh saha khoya dziri wash kash jdid rani mezelt 7asl m3a limei

abassi94, 2016-12-22:
Araw7o nbi3lk 
System ta3 imei hhhhh

alilihicham, 2016-12-23:
abassi94  arwa7 llfb ne7tajek 
https://www.facebook.com/hicham.h.mahcih

MrHackmed, 2017-01-18:
@chipoza Can you please upload your nvdefault.img file to the internet and share the download link here then, @forth32 can explain to us better on how to restore it to our devices.

closed 2017-09-24 by forth32