forth32/balong-usbdload #9: e5573cs-322 help

HUAWEI-MODEM, 2016-11-02:
дорогой у меня есть e5573cs-Huawei 322 Устройство ваш загрузчик не работает на этой модели у меня есть Orignal загрузчик для этого устройства у просто залатать его .when я применить этот загрузчик, то Nv данные, как номер IMEI seriel все Vipe kindely залатать этот загрузчик

https://drive.google.com/file/d/0B8ZHTO36GnaSc2EyTndnOVdZZ1k/view?usp=sharing

hassansagar78, 2016-11-03:
after this loader modem only support downgrade firmware 21.318.99.xx.xx and no wifi or Rndis ..the modem didnt came into working position or support upgrade firmware 21.318.03.00.00 ?
(modem e5573cs-322 zong cmpak)

HUAWEI-MODEM, 2016-11-03:
forth32 please patch this loader we are waiting for your response

forth32, 2016-11-03:
This is no more need to patch loader. This program do it on the fly, during loading.
Or, you can patch loader by program loadr-patch.

hassansagar78, 2016-11-03:
after the loader downgrade 21.318.99.00.00(normal) succesfull..modem dual light always on  but when try it upgrade 21.318.03.00.00 (general) it says the wireless terminal is not e5573cs ..when flash 21.318.03.00.00 with balong_flash utility, it completed but modem did not turned on and went to dead mode (download port ) ..
no upgrade firmware supported 
(e5573cs-322 zong cmpak)

HUAWEI-MODEM, 2016-11-04:
dear this loader is not patch on fly when i run this loader all data like imei seriel mac adress firmware webui all vipe kindly i request to you please patch this loader we will thanks 2 u.

forth32, 2016-11-04:
You are not right:

```
forth32@acer ~/huawei/balong-usbdload $ ./balong-usbdload usbloader-5573cs-322.bin 

 Аварийный USB-загрузчик Balong-чипсета, версия 2.02, (c) forth32, 2015

 * Удалена процедура flash_eraseal по смещению 00000460

```

Use last version of this program (build it from sources). You can patch ythis file permanently by:

./loader-patch -o usblsafe-5573cs-322.bin usbloader-5573cs-322.bin

HUAWEI-MODEM, 2016-11-04:
cammand is not work

hassansagar78, 2016-11-04:
forth32 Sir
kindly patch this  loader .
usbalsafe-e5573cs.bin  didnt work for this model as it was e5573cs-323 loader.bin..
kindly make patch for this loader e5573cs-322

moizzzz, 2016-11-04:
same problem with me after trying usbloader.bin all data suck .after flash no reset with reset button but i am trying with nv read writer tool .so dear patch this loader for all users

nomiyousafzai, 2016-11-04:
Dear all,
i don't think safe loader will work because main issue is re update on general 21.318.03.00.00.
every thing is going right problem is when update on general version.

hassansagar78, 2016-11-04:
yes man .No upgrade firmware supported by this model currentely...whenever flash upgrade 21.318.03.00.00 it went back into dead mode only 1 boot pin port

nomiyousafzai, 2016-11-04:
yes i am also facing this issue.
Russian version 21.318.03.00.778 for e5573cs-322 also having same issue.

HUAWEI-MODEM, 2016-11-05:
dear first we need usblsalf.bin patch loader then next we decide fw is wrong or have any other security in this model.

HUAWEI-MODEM, 2016-11-05:
forth32 sir please patch this loader please

hassansagar78, 2016-11-05:
patch it on fly balong_usbdload latest build but it never comlete execution and also not wipe any data ?
zong cmpak modem have new type of security

forth32, 2016-11-05:
I do not know what i can patch in this loader in addition to autopatch procedure in balong_usbdload and loader_patch. This patch fully removes flash_eraseall procedure from loader.
I do not have this router and can not test loader on real hardware. Try to solder wire to hardware UART sender pin (TxD), connect it to USB-UART adapter and view log during usbloader boot process. This log is needed to determine problem.

Ptobably, this loader is designet for other hardware, not 5573-cs.

hassansagar78, 2016-11-05:
yes sir you are right..
patched on fly sucessfull .after it downgrade firmware is accepted.
no data wiped during the process..

nomiyousafzai, 2016-11-05:
hassan kindly share real command used.

nomiyousafzai, 2016-11-05:
Hassan what s progress????
Updated on 21.318.03.00.00 or not???

hassansagar78, 2016-11-05:
amazingly not

hassansagar78, 2016-11-05:
Forth32 sir 
how to flash 19 .OOB Files ?which i dumped from another working modem

nomiyousafzai, 2016-11-05:
loader writes successfully but device in in download port 
C:\Users\nomi\Downloads\Programs>abc -p53 usb.bin

 ????????? USB-????????? Balong-???????, ?????? 2.02, (c) forth32, 2015
 ???? ??? Windows 32bit  (c) rust3028, 2016

 * ??????? ????????? flash_eraseal ?? ???????? 00000460

 ?????????    ?????    ??????   %????????
------------------------------------------
 raminit    10000000     3936   100%
 usbboot    57900000  5844136   100%
????? ?????? ????? ????? ??????

 ???????? ????????

HUAWEI-MODEM, 2016-11-05:
thanx forth32 thank u very much

nomiyousafzai, 2016-11-05:
successful or not?

hassansagar78, 2016-11-05:
downgrade success already.. dont know which upgrade firmware would support ?
every firmware  ends with putting modem in dead mode again.

nomiyousafzai, 2016-11-05:
i used loader all nv wiped, i removed battery and restart modem but now only one port  download port mode.
now??

hassansagar78, 2016-11-05:
same problem with every firmware except 21.318.99.03.00

hassansagar78, 2016-11-07:
1- usblsafe-e5573cs-322 by rust3028 ,tried with balong_usbdload latest build as well as with balong usb downloader but device didnt turned .ending  with error
Log Is Below

Ïîèñê óñòðîéñòâ:

COM5 - HUAWEI Mobile Connect - Control Interface [BOOT_3G]
        USB\VID_12D1&PID_1443&REV_0001
        USB\VID_12D1&PID_1443

Ïàðàìåòðû çàãðóçêè:
00000001. 00000F60 10000000 00000054
00000002. 00592CA8 57900000 00000FB4
Íàéäåíî 2 áëîêà(îâ) äëÿ çàãðóçêè.

Ïûòàåìñÿ îòêðûòü óñòðîéñòâî \\.\COM5 ...
Çàãðóçêà áëîêà 0, àäðåñ=10000000, ðàçìåð=3936
Âñåãî ïàêåòîâ â áëîêå [0]: 4
Çàãðóçêà áëîêà 1, àäðåñ=57900000, ðàçìåð=5844136
Âñåãî ïàêåòîâ â áëîêå [1]: 5708
Ìîäåì îòâåðã ïàêåò êîíöà äàííûõ. Îáðàç ïàêåòà:

0000: ED 4D B2 1E 69                                   *íM..i           *

nomiyousafzai, 2016-11-07:
what is going on every thing failed no 2nd option only zong firmware is writing successfully.

hassansagar78, 2016-11-07:
@nomiyousafzai 
can you provide link for zong firmware e5573cs-322?
(hassansagar78@gmail.com)

HUAWEI-MODEM, 2016-11-07:
send me zong fw link (alivekhanpk@gmail.com)

huaweihack, 2016-11-07:
Dear nomiyousafzai plz provide me e5573cs-322 zong firmware at heartless123ravi@gmail.com thanks alot

nomiyousafzai, 2016-11-07:
i heared i am also looking for it.

nomiyousafzai, 2016-11-08:
BAD NEWS E8372H-153 
21.318.01.02.306
NOT ACCEPTING SAFE AND ORGNL LOADER.
THAT MEANS UNLOCKING FINISH.

capttin, 2016-11-16:
plz firmware send me zong e5573cs-322 capttin.mobile.center@gmail.com

capttin, 2016-11-16:
plz send me new e8372h usbloder capttin.mobile.center@gmail.com

capttin, 2016-11-23:
frends click this link but not free http://www.huawei-file.com/index.php?dir=/E_Series/E5573Cs-322

shah-95, 2017-01-13:
hello , is there any solution for e5573Cs 322 zong pak? it does not accept upgrade firmware 21.318.03.00.00 , any solution for it? can anyone give me working zong firmware ?

Mohsan123, 2017-02-01:
how to copy bootloader from mifi?

iabros, 2017-02-04:
Forth32 plz give us picture/snapshot dumping fw from device

Rimtopak, 2017-02-10:
Hello ,
     Any updates pls

iabros, 2017-02-10:
No my friend

Rimtopak, 2017-02-11:
Hello DO some needful pls

Mohsan123, 2017-02-17:
@forth32, is there any solution for e5573Cs 322 zong pak.

hassansagar78, 2017-02-28:
now enjoy happy unlocking for this model

Rimtopak, 2017-02-28:
How dear ??

Rimtopak, 2017-02-28:
share the updates pls

Rimtopak, 2017-02-28:
Did you find the solution pls

iabros, 2017-02-28:
Plz share with us

Rimtopak, 2017-02-28:
hassansagar78 pls share the solution if you have ??

iabros, 2017-02-28:
I know hassan say truth hassan brother plz share us request u

closed 2017-02-28 by forth32

Rimtopak, 2017-02-28:
closed but why??

iabros, 2017-02-28:
Forth close because I know this solution made forth and only give some
selected people in pakistan

Rimtopak, 2017-02-28:
Pls how to get this solution tell me pls

Rimtopak, 2017-02-28:
i have only this one device . Pls support me

iabros, 2017-02-28:
Brother plz read my reply I have no solution I am telling u truth I request
many forum but no buddy give me solution plz trust me I also request u I u
got solution plz give me

Rimtopak, 2017-02-28:
Dear Iabros I know that you have no solution. Thanks for reply.

shoaibrao, 2017-02-28:
@hassansagar78
Sir You Got Solution ?