forth32/balong-usbdload #10: e8372h loader problem

chipoza, 2016-11-09:
I am not able to lower my e8372h-153. 
operator Zong CMPak Pakistan. 
firmware version 21.318.01.02.306. 
safe and original loader also does not work and gives an error 
link is also attached.
http://4pda.ru/forum/index.php?showtopic=678549&st=1400#entry54916395

chipoza, 2016-11-09:
in this firmware version 21.318.01.02.306 partition size change from 5.8mb to 5.3 mb.and active kernal when we use these  loader usbloader.bin or usblsalf.bin then kernal restore nv ram.
solution ;>
you patch loader who deactivate firstly kernal and after boot write

hassansagar78, 2016-11-13:
e8372 21.318.01.02.306 not accept any loader now on this version 
need to find a new loader

naeemkkk, 2017-02-27:
Same Problem With Me :( My Zong E8372h-153 Not Loading Usblsafe.bin File But How Can I Solve It Please Experts Help Me :(

Mohsan123, 2017-03-01:
@naeemkkk  use e5573cs-usbloader.bin . this will work.

naeemkkk, 2017-03-02:
@Mohsan123 bro this file is loaded but firmware is not update any soulation please do you have any firmware working  please suggest me bro.

Mohsan123, 2017-03-03:
yup! I have working firmware. fw 21.318.01.02.306. problem is nvram clear after updating by this firmware.

naeemkkk, 2017-03-03:
@Mohsan123  Please can you give me the firmware?

Mohsan123, 2017-03-03:
yes offcourse.

balwinder27, 2017-03-03:
Forth32 and Mohson 123 both are greedy persons 
I put comment greedy forth32 locked the thread 
Shame on you both peoples

balwinder27, 2017-03-03:
Without usbloader forth 32 is zero 
If you are too programmer then make a 1 usbloader 

Just copy loaders from huawei servers then just patch from huawek database 
Then just modify a firmwares 
Forth32  don't think you are no1 because without huawei loader  you are zero

wsw243, 2017-03-03:
who want zong E5573cs-322 solution contact me on skype WSW243

closed 2017-03-03 by forth32