forth32/balong-usbdload #11: E3131i-2

Terminator8, 2016-12-02:
Hello
Can you find usbloader.bin for this model or perhaps "how to" for retrieval by firmware or modem that works?
Thanks in advance.

chipoza, 2016-12-02:
i can not find this model usbloader.bin you try all model usblsalf.bin in forth32 project and try with this.maybe any loader work.

forth32, 2016-12-02:
E3131 is Balong V3 based modem, not Balong v7. I am not sure that this program can download usbloader for V3 chipset. And do not try usbloaders from this repository - these only  for Balong V7.

Terminator8, 2016-12-02:
Thank you both for the quick response.

balwinder27, 2017-03-04:
https://drive.google.com/file/d/0B3pEUGa_GFBrS01XLU4zM2I1RHc/view

e8372h-153 zong free solution for everyone

closed 2017-09-24 by forth32