forth32/balong-usbdload #12: e8372h-153 loader problem

chipoza, 2016-12-04:
dear sir i creat full video for e8372h-153 loader problem
https://drive.google.com/file/d/0B8ZHTO36GnaSMU1NX0RwM21CaDg/view?usp=sharing

forth32, 2016-12-04:
To determine problem source, i must view log from internal hardware UART. Solder USB-Serial converter to TxD pin (located on main board of router), open any terminal program with parameters 115200-8-N-1 and grab log during usbdload loading.

hassansagar78, 2016-12-04:
http://imgur.com/gallery/0VEsx
 where is its TxD pin kindly mention back

forth32, 2016-12-04:
One from 4 pins (usually +5v, GND, Rx, Tx). Use oscilloscope to determine Tx pin. You can also probe all 4 pin one-by-one. Connecting each of them to serial converter Rx terminal and see log.

chipoza, 2016-12-04:
i have not oscilloscope .you can show me +5v gnd Rx Tx.
https://drive.google.com/file/d/0B8ZHTO36GnaSbTk0TDBJRm9oalk/view?usp=sharing

forth32, 2016-12-04:
I can not show these pins - i do not have this device.
Use multimeter to determine GND and +5V. Two other pins will be Rx and Tx.

chipoza, 2016-12-06:
i find these point picture is below so what next
https://drive.google.com/file/d/0B8ZHTO36GnaSSFZHMXBOV0h6RU0/view?usp=sharing

chipoza, 2016-12-07:
i sold the four point .and then i insert in pc .it show error usb not recognized.if you need sample then i can send to you.

iabros, 2017-02-17:
forth I know u develop new firmware .fw file for this wingle plz share us

naeemkkk, 2017-02-27:
Same Problem With Me Please Update And How To Unlock New Firmware E8372h-153

balwinder27, 2017-03-03:
i sucessfully unlock done a zong e5573cs-322 and e8372h-153 new logo

naeemkkk, 2017-03-03:
@balwinder27  Brother, can you tell me this method. My E8372h-153 has been dead not accept any firmware if u share your method.

Mohsan123, 2017-03-03:
@naeemkkk  I give you this method free of cost.

iabros, 2017-03-03:
Mohsin give me e5573cs-323 plz

Mohsan123, 2017-03-03:
@iabros  I said I will give  E8372h-153 method not e5577cs-323.

iabros, 2017-03-03:
Why u not give e5573cs-322 solution u don't lie u have solution I know

balwinder27, 2017-03-03:
Iabros bro I give you 

Both solutions free e8372h-153 Zong new logo 
Zong e5573cs-322

Contact on my ICQ
+919646599249

I give you solution free

balwinder27, 2017-03-03:
I have also e5573cs-323

balwinder27, 2017-03-03:
E5573cs-933 free solution avaiable

closed 2017-03-03 by forth32