forth32/balong-usbdload #14: nvram restore

abassi94, 2016-12-20:
hi forth32 
i read your analysis off the fast boot problem you said the nvram is automatically restored 
but i in my b310s 927 it does not 
it stays empty i have lost imei annd serial  number is there any way i can restore it 
thnx

The-BigNoob, 2016-12-21:
i have found this file 
https://github.com/forth32/balong-usbdload/blob/master/usbloader-b310.bin
with  this firmware ( B310s-927TCPU-V200R001B110D99SP02C00_Firmware_general_05021QJS ) 
the same file 
so this usbloader is patched or not ?? because all nv data  wiped. i backup only (mnvm2:0/nv.bin)
sir forth32 we need help plz

chipoza, 2016-12-21:
you can fix this manually
for imei repair
at^datalock="12345678"
AT^cimei="15 digit imei"
for seriel no. repair
It is necessary to translate the serial number string into a sequence of codes and written in the NVRAM AT-command.
For example, the serial J5HBYA93C3042173 should be written as follows:
AT ^ DATALOCK = "unlock code"
the AT ^ = NVWR 6,20,4A 35 48 42 59 41 39 33 43 33 30 34 32 31 37 33 FF FF FF FF

forth32, 2016-12-21:
To enter new serial number, yuo can use command

at^sn=serial

chipoza, 2016-12-21:
bro can we repair mac adress?

The-BigNoob, 2016-12-21:
nothing happen it's impossible to repair imei and serial number with AT command

Oem and Nck on my device ALL 000000000

^NVRDEX: 50502,0,128, 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

^NVRDEX: 50503,0,128, 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

sir forth32 if usblsafe repair Oem and Nck please make usblsafe for B310s
or any other method 

thank you

abassi94, 2016-12-21:
Big noob can you upload the 
Backup

abassi94, 2016-12-21:
Forth can you complie the patch tool for 
Windows .
I could not run in linux

freereial, 2016-12-22:
Hello .. how to solve the light power in modem b310-927 What is the update user programs .... Please help me

chipoza, 2016-12-30:
contact to me i will solve all user problem.only b310-927 //// icq uin 704208779

hassansagar78, 2017-02-12:
if we repair manualy imei and serial number it works fine but the problem is device didnt reset anymore...
how to fix reset issue manualy except restore nv?

brightosky27, 2017-02-13:
b315 safe loader works for b310 as well

brightosky27, 2017-03-03:
> @forth32 <https://github.com/forth32> please guide me for "at^" command
> that change SSID & Wifi password in NVRAM. I wants that it never change
> after reset device.

i dont get explain again

forth32, 2017-03-04:
Firmware from balwinder27 requires mkey dongle (it is in mkey .cbin format), therefore, this is mkey marketing. In this reason, i will block all balwinder27 messages in future.

Don not convert this issue threads to market trash. Use mkey forums fo it.

closed 2017-09-24 by forth32