forth32/balong-usbdload #15: lost imei and serial number (B310s)

The-BigNoob, 2016-12-21:
hi sir forth32
i lost imei and serial number after flashing and i have backup only this file (mnvm2:0/nv.bin)
How do I restore nv data?
thank you

chipoza, 2016-12-21:
for imei repair
at^datalock="12345678"
AT^cimei="15 digit imei"
for seriel no. repair 
It is necessary to translate the serial number string into a sequence of codes and written in the NVRAM AT-command. 
For example, the serial J5HBYA93C3042173 should be written as follows:
AT ^ DATALOCK = "unlock code" 
the AT ^ = NVWR 6,20,4A 35 48 42 59 41 39 33 43 33 30 34 32 31 37 33 FF FF FF FF

abassi94, 2016-12-21:
Can i have the backup pls

The-BigNoob, 2016-12-21:
nothing happen it's impossible to repair imei and serial number with AT command

Oem and Nck on my device ALL 000000000

^NVRDEX: 50502,0,128, 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

^NVRDEX: 50503,0,128, 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

sir forth32 if usblsafe repair Oem and Nck please make usblsafe for B310s
or any other method

thank you

chipoza, 2016-12-22:
bro if you oem is zero you can set your oem to 99999999 by this method
To do this, you need to read the contents of the cell 50502 NVRAM, 
^ = NVRDEX the AT 50502,0,128 
^ NVRDEX: 50502,0,128,63 the A7 19 93 3F 65 CF 07 18 1B AC 0A C8 C2 4B 0C 9C 4D 28 34 73 E1 15 79 6B 12 EB 1D 84 15 58 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 A6 22 61 59 4B AD BF 74 AA FD 02 6C BA 22 C3 F8 78 EC 9F A2 3D D1 EF F0 D8 A8 C0 D5 63 36 C3 07 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

OK

your modem codes will be different) 
, replace the last non-zero (01) on 00 and write everything back.

AT ^ NVWREX = 50502,0,128,63 A7 19 93 3F 65 CF 07 18 1B AC 0A C8 C2 4B 0C 9C 4D 28 34 73 E1 15 79 6B 12 EB 1D 84 15 58 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 A6 22 61 59 4B AD BF 74 AA FD 02 6C BA 22 C3 F8 78 EC 9F A2 3D D1 EF F0 D8 A8 C0 D5 C3 63 36 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Or you can do it in conjunction with the OEM-code setting in 00000000:
AT ^ NVWREX = 50502,0,128,8F 29 FF 8E A8 CA 34 89 78 73 18 BA 9E F5 9C 64 B A4 DB 81 DC 6E 3 45 72 DA EC 6A C 7C 90 65 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 8B 8C F4 B5 AF C F2 2C FE E0 F4 46 9C CF 47 95 36 71 1F 1C BF 5 7F 84 AB A9 F2 92 3C 89 33 12 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
after at^reset.
then for imei repair
at^datalock="12345678"
AT^cimei="15 digit imei"
for seriel
at^sn=12345678910

The-BigNoob, 2016-12-22:
sir chipoza 
I've tried you method and all ways but without success 

AT^NVRDEX=50502,0,128  is Empty all NVRAM records wiped
^NVRDEX: 50502,0,128, 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

chipoza, 2016-12-30:
dear can you check me through teamviewer?

bezza06, 2017-02-12:
The-BigNoob change the card sim and send the command 

at^datalock="12345678"
AT^cimei="222222222222222" 16 digits
AT^mode=0
AT^reset

try this for several times

balwinder27, 2017-03-04:
https://drive.google.com/file/d/0B3pEUGa_GFBrS01XLU4zM2I1RHc/view

e8372h-153 zong free solution for everyone

closed 2017-09-24 by forth32