forth32/balong-usbdload #17: Please Help Patching E5577 usbloader Link inside

Zaraky2008, 2017-01-29:
I've not seen patched usbloader for E5577
Please Help Patching E5577 usbloader 

and can anyone made usblsafe from it ?

Link to it 

https://archive.org/download/E5577Usbloader/E5577_usbloader.rar

forth32, 2017-01-29:
There is no need to create usblsafe file now. Currently, loaders will be patch on-the-fly, in loading process, by balong-usbdload utility.
Ypu can safety  load standart unpatched  usbloader.

Zaraky2008, 2017-01-29:
I used your  Balong USB Downloader 1.0.1.10 program 
but after the program finished 
I cannot use other upgrade than the one version 21.180.99.03.00

If I used the forth32/balong-usbdload using windows CMD the usbloader will be patched by it
and I can run other upgrade version like 21.200.09.00.00 after finished

Zaraky2008, 2017-01-29:
the usbloader in the link came with the firmware version 21.180.99.03.00
after I used it with Balong USB Downloader 1.0.1.10 program 
the device cannot accept any firmware other than 21.180.99.03.00

If I used the forth32/balong-usbdload using windows CMD instead of Balong USB Downloader 1.0.1.10 program the usbloader will be patched by it
and I can run other upgrade version like 21.200.09.00.00 after finished

Zaraky2008, 2017-01-29:
???????????????????????????????????????????

Zaraky2008, 2017-01-29:
Thank you very much Mr.forth32 for your reply

Zaraky2008, 2017-01-29:
is Balong USB Downloader 1.0.1.10 program can patch the usbloader.bin 

or I use balong-usbdload with cmd by it the usbloader.bin will be patched by it

forth32, 2017-01-29:
balong usb downloader is not my program. It is fake, created by Decker from my sources with GPL violation. Do not use this program.

Zaraky2008, 2017-01-30:
Mr. forth32 I had a problem with the same usbloader.bin
I used your balong usbload with port 10 after it run and finished 100% 
The router wifi E5577 does not restart and goes into 2 ports for downloading firmware. The device still off and it still stuck in boot shot one port which is port no.10

but when I used  Balong USB Downloader 1.0.1.10 program by Decker to run usbloader.bin
The device restart and show "installing" in same device screen  and goes into 2 ports for downloading firmware.

I'm confused ??

Can you make patched or usblsafe out of E5577 usbloader.bin I sent to you ?

pleaase kindly

chipoza, 2017-03-01:
i think you need to sign firmware+webui in 1 file.it will work .

Zaraky2008, 2017-03-01:
How can you teach me ?

Zaraky2008, 2017-03-01:
I've all the necessary firmwares in exe and bin 

I just want guidance and program to merge the update+webui into one binary file

pleaaaase help

balwinder27, 2017-03-04:
https://drive.google.com/file/d/0B3pEUGa_GFBrS01XLU4zM2I1RHc/view

e8372h-153 zong free solution for everyone

closed 2017-09-24 by forth32