forth32/balong-usbdload #20: Zong CmPak E5573Cs-322 .

BKHNS, 2017-03-10:
@forth32: 
Your assistance is required with regard to **E5573Cs-322** having a **signed** firmware version **21.318.03.01.306**. 
The loader helped in erasing NV contents and then the device downgraded successfully to 
21.318.99.03.00, i was also able to restore the NV contents later but after upgrading to 21.318.03.00.00, the device remains in dead mode. 
Please help me out if you find it appropriate, thankyou.

wasifmobileshop, 2017-03-13:
hi
E5573Cs-322 unlock don 2g signal not 3g and 4g signal ok pless sir solve my problem E5573cs_UPDATE_21.318.09.00.00 thinks

wsw243, 2017-03-13:
@BKHNS 
contact with me on skype WSW243

forth32, 2017-03-13:
Please do not convert this issues to commercial forum!!!

chipoza, 2017-03-17:
dear @wasifmobileshop @BKHNS forth32 is working on e5573cs-322 .it will gave solution to people as soon as possible.and thanks  again forth32 for e8372h unlock.

BKHNS, 2017-03-20:
Thank you @chipoza for the reply.

chipoza, 2017-03-20:
according to forth32 some pakistani people gave e8372 solution to mkey admin . mkey add this fw in our support area .due to this fort32 is not reply pakistani people.but that is not true .india guy blwinder27 leack this fw.

Mohsan123, 2017-03-22:
according to e8372h solution by hassansagar and blwinder27, all nvram data will be wiped. so it is not safe solution as compare to forth32 provided solutions.

Mohsan123, 2017-03-22:
according to e8372h solution by hassansagar and blwinder27, all nvram data will be wiped and this is also not free. so it is not safe solution as compare to @forth32 provided solutions that's free of cost.

shah-95, 2017-03-23:
@chipoza what was the solution for e8372h-153 unlock ??

chipoza, 2017-03-23:
we have safe solution but this solution is not shareable due to market reason . solution created by my teacher forth32  .(world first time)

shah-95, 2017-03-23:
if it is free then why not share it bro , i am not any seller or repairer , i want only to make my own device working , if you cannot share the solution , then you and others that charge hefty amount are the same , sorry to say bro this is not the good way of life

chipoza, 2017-03-23:
the owner of this solution is forth32 . i can not leak without fort32 permission . forth32 will release final version for e8327 .in this time have some bugs mac address etc. if you want own device in working you just 
boot pin and load usbload.bin and simply run e5573cs-322 orignal fw. after update your device will be in working condition .

shah-95, 2017-03-23:
ok dear , no problem , i already know this , just wanted a patched firmware for working with other networks bro , i believe in this that if you get helped for free , help others for free , if you ever could give me patched fw it would be much kind of you , regards.

chipoza, 2017-03-23:
This is helping project (no paid like mkey). forth32 will release e5573cs-322 solution but wait some time because have some bugs.i know some people created solution but this solution is not full also have some bugs . 2g not work , no stability , reset button issue , etc.

shah-95, 2017-03-23:
@chipoza yes u are absolutely correct , i am having the same issues bro , i want your help bro , can you contact me at ali_hassan_shah19@yahoo.com , thank you very much

renoos, 2017-04-09:
hi to all .. i am here new . and first i want to thank you sir forth32 for making free solution .

shah-95, 2017-04-11:
can this FREE solution not be shared ??? why we had to pay heavily to some who get this solution for free ???

chipoza, 2017-04-11:
This is only a helping section . We not demand pay .

Mohsan123, 2017-04-11:
We are waiting solution for e8372h-153, e8372h-608 & e5577cs having latest firmware,
We hope Great Sir Forth32 will provide as soon as possible.
someone who have this solution is not complete solution.

chipoza, 2017-04-11:
We can not leak solution due to mkey marketing .it has some -ve rezult .if we leack openly then operater will update software and other due to market reason we can not show this method openly . If you want to unlock this device need to learn more and judge the security  and more .otherwise we can no performer any action without forte32 permission .if you have any question contect with me alivekhanpk@gmail.com

BKHNS, 2017-04-11:
@forth32 , KIndly grant that (PERMISSION) that everybody on every form keeps talking about. 
If possible kindly contact me at: burhankhans@outlook.com. 
**I don't need your solution for any profit motive, the only reason is to restore my device into working condition.** 
Pls respond if appropriate.

hassansagar78, 2017-04-13:
@mohsan123 bro stop blaming me..if you have any personal issue than email me hassansagar78 @gmail.com and for your kind information I hv both safe solutions zong 5573cs-322 and zong 8372h-153

chipoza, 2017-04-13:
@mohsin Please dont use immortel language otherwise you Will we banned. As
à rezult forte32 can not help to pakistani Person .please do not use bad
language request to all .

tanveerahmad6506, 2017-04-16:
zong green logo unlocking 1000% ok...
skype id: tanveer6506@hotmail.com

closed 2017-04-16 by forth32