forth32/balong-usbdload #21: e3372h-153 console problem after usbload/flash

bjne, 2017-03-11:
It is no longer possible to write AT-commands into the serial port, and seems the mode is
set permanent to some write-mode, this is what I see:

balong-usbdload# ./balong-usbdload -p /dev/ttyUSB3 usblsafe-3372h.bin

 Аварийный USB-загрузчик Balong-чипсета, версия 2.02, (c) forth32, 2015

 * Удалена процедура flash_eraseal по смещению 000004b0
 ! Порт не находится в режиме USB Boot
root@roadwarrior:~/e3372h-debug/balong-usbdload# socat - /dev/ttyUSB3,crlf
A
2017/03/11 03:08:05 socat[567] E write(5, 0x256e930, 2): No space left on device

..this is actually after trying lots, and ending up with doing dd of /dev/zero into ttyUSB3

the webui is still working, tho

forth32, 2017-03-11:
This program is only for boot usbloader. It is not AT or seril console.
Before using it, modem must be put into usb-boot mode.

bjne, 2017-03-11:
I may not have explained myself good enough. The following was what happened:

1) I started e3372h-153 in fastboot mode
2) ./balong-usbdload -p /dev/ttyUSB3 usblsafe-3372h.bin
-> success, device entered loader mode, with rapidly blinking white/blue light
3) ./balongflash -p /dev/ttyUSB5 E3372h-153_Update_22.323.01.00.00_M_AT_05.01.bin
-> failure:  Неправильная версия монитора прошивки - [5]

Ok.. checking the sourcecode shows me the problem is caused by reply from the
AT^DLOADVER? command, so I want to check the output of this command myself,
connecting to /dev/ttyUSB5: socat - /dev/ttyUSB5,crlf

The output surprises me, it sends me "AT^DLOADVER?", so I try sending ATI,
result is it closes the connection. Trying again, this time when I start socat, I get:
"AT^DLOADVER?ATI"

hmm..

So I disconnect the device, and try entering fastboot again. This time balong-usbdload
fails.

Device "seems" to work in normal mode, (atleast ethernet device and webui is up), but the
modem port is acting strange still, and to my surprise it still sends me the "AT^DLOADVER?ATI",
and whatever else I send into it..

after a while trying different things and modes, I descide to check what will happend if I do
"dd if=/dev/zero of=/dev/ttyUSB3"

it writes until eof, and now I am in this state:

# socat - /dev/ttyUSB3,crlf
A
2017/03/11 03:08:05 socat[567] E write(5, 0x256e930, 2): No space left on device

(the high ttyUSB numbers is caused by a integrated qualcomm 4g modem using 0-2)

closed 2017-09-24 by forth32