forth32/balong-usbdload #24: Huawei E3531 загрузчик

snegin, 2017-03-20:
Пожалуйста добавьте загрузчик под этот модем.

forth32, 2017-03-21:
А где его брать-то?
Это только если выкинуть кучу денег и перерыть весь asanfile, в надежде найти его в каком-нибудь из файлов прошивок. И потом, это же 3G модем, не на Balong V7. Неизвестно, поддерживает ли он такой протокол загрузки.

snegin, 2017-03-21:
Скорее всего не поддерживается протокол.Пробовал заливать софт одной из Ваших программ - пишет "неизвестный протокол".Вопрос снимается.

forth32, 2017-03-21:
Вообще-то программа balong-usbdload не умеет писать "Неизвестный протокол". Она умеет только сообщать "Порт не находится в режиме USB Boot".
Так что тут еще вопрос, той ли программой делались попытки загрузки...

snegin, 2017-03-21:
Фирмваре заливал не balong-usbdload,а другим Вашим софтом для Хуавеев. balong-usbdload заливал загрузчик от другой модели,но с одинаковым процессором.Если отбросить процессор(такой же),протокол(возможно он один) в чём отличие между разными моделями?Память?Или каждый загрузчик под конкретную модель?

forth32, 2017-03-21:
Как правило, в пределах одного и того же SOC все загрузчики взаимозаменяемы. То есть, например, к Е8372 подойдет загрузчик от 3372Н. Но если в модемах отличается объем встроенной памяти, то может не подойти первый компонент загрузчика - raminit. Он собирается под конкретную карту адресного пространства, а она может и отличаться. Например, хагрузчик от В315 не подходит к Е3372Н, хотя они оба сделаны на одном и том же чипе Balong V7R11.

snegin, 2017-03-21:
Спасибо что разъяснили.
"Модем отверг пакет конца данных" - такой ответ при попытке загрузки загрузчика.

forth32, 2017-03-22:
Это как раз говорит о том, что протокол загрузки здесь такой же, а вот адрес загрузки, скорее всего, неправильный. Нужен другой загрузчик.

snegin, 2017-03-22:
Не один из выложенных не подходит.Искать я так понимаю в прошивках?

forth32, 2017-03-22:
Да, конечно. Некоторые прошивки с asanfile содержат в себе usbloader, но тратить деньги на слепой поиск нужного файла у меня, например, нет никакого желания.

snegin, 2017-03-22:
Это понятно.Поищу я.Что в нем должно быть обязательно?На что обратить внимание.Пример кода.

forth32, 2017-03-22:
Он должен обязательно называться usbloader.bin - других имен загрузчиков я у хуавеев не встречал.
Ну, а внутри должно встречаться слово "raminit" например.

snegin, 2017-03-22:
Его нужно вырезать из кода или он распаковывается из ехе?

forth32, 2017-03-22:
Он лежит в архиве с прошивкой отдельным файлом.

chipoza, 2017-04-16:
sir if you need any file from asanfile or dlgsm . i can give it just give me link i will upload for you.........

snegin, 2017-04-16:
I need any firmware zip which contains usbloader.bin from http://asanfile.com/file/index/file?path=huawei%5Cflash%20file%5Ce%20series%5Ce3531s%5C.Thanks.

snegin, 2017-04-16:
http://asanfile.com/file/view?fileName=huawei%5Cflash%20file%5Ce%20series%5Ce3531s%5Ce3531s-2tcpu-v200r001b316d99sp16c00_general_05021qvx.zip

chipoza, 2017-04-16:
https://drive.google.com/file/d/0B6zT9PC9ED4XRmY4Q3RwZUJ4Q1U/view?usp=sharing

anix22, 2017-04-30:
https://drive.google.com/file/d/0B3pEUGa_GFBrcDdxU1FFc2RoRGs/view

ZONG E8372H-153 RESET FIXED SOLUTION SPEHIAL THANKS TO VICKY AND nadeem danish

chipoza, 2017-04-30:
@anix22
forth32 will be block to you for your marketing.

ronix889, 2017-05-06:
chipoza sir can you tell bhow to use balong nvtool.exe?

ronix889, 2017-05-07:
any one help????????????

closed 2017-09-24 by forth32