forth32/balong-usbdload #26: E8372h-608 with fw 21.316.01.04.274

Mohsan123, 2017-04-11:
Huawei wangle e8372h with latest firmware 21.316.01.04.274 no more support old loader. any solution?

chipoza, 2017-04-11:
Mohsin contect with me. Alivekhanpk@gmail.com

eriksejr, 2017-04-12:
I have the same problem. I have a bunch of older 8372h-512's and some newer e8372h-517's. balong-usbdload & the usblsafe-8372.bin loader work perfectly on the older 8372-512's. On the newer e3872-517 devices the usbdload program sends the loader but they don't appear to properly execute the loader. The two ttyUSB ports appear. but the LED does not start flashing and balong-flash cannot send firmware to them. I soldered on to the serial console port of the device as well, and i've attached the boot output (including the partition table) of the newer 8372h-517 device. Attempts to upgrade/downgrade the newer devices using the "official" firmware installers also fail. I cannot find any firmware image anywhere for the e8372-517 - this appears to be a fairly uncommon submodel of this device.

 As I cannot downgrade the firmware on this device I do not have the OEM code for it so I cannot gain console/root access to it. I have been unable to gain access to the NVRAM to get the hashes to calculate the codes...

[wingle.txt](https://github.com/forth32/balong-usbdload/files/917156/wingle.txt)

chipoza, 2017-04-12:
If your device is newer version. You need to signed loader(huawei secret
key verification ) or oem coad . You can unlock this device with
usbloaders.bin (unsafe) . Load usbloader.bin through balong utility after 2
ports will be appear .run go_hdlc next open huawei v2.0 flasher and simple
enter comm port number and flash it .after flashing device will be run in
normal mode . If device go to boot point then you need orignal signed fw
for this device .you can not access adb (telnet) without oem coad

shakil01, 2017-04-13:
hey chipoza same problem with my e8372-153 usblsafe-8372.bin or usbloader.bin not working any solution for this ..........

renoos, 2017-05-02:
@Mohsan123 @chipoza @forth32 
Huawei wingle e8372h with  firmware 21.316.01.03.274 unlocking solution plz ?

Mohsan123, 2017-05-17:
Huawei E8372h-608 having firmware 21.316.01.04.274 (Operator Telenor)
Resettable Unlock Done! by using Respected Sir @forth32 ideas.

Thanks So Much Respectable Sir Forth32.

closed 2017-05-17 by Mohsan123