forth32/balong-usbdload #27: e5172 dead  please help forth2 sir

anix22, 2017-04-16:
hi sir my  e5172s-22 huawei modem dead 

i find 1 firmwrae which contain usb loaders and balong files
balongv7r1_bootrom.bin
balongv7r1_mcore.bin
bootloader.bin
fastboot.bin
ptable.bin
si32178.ko
usbloader.bin
zImage

sir how to write these files on router
i use balong tool on e5172 and use usbloader
device make 2 com ports pc ui and application ports
i try write firmware on e5172 bin but it fail to write

sir help me please my modem is  not work

chipoza, 2017-04-16:
SOLUTION=
1>load usbloader.bin after 2 comport will be appear .
pc ui interface and application interface ..
if you are linux user loader usbloader.bin and it show device names /dev/ttyUSB*
2> download this fw https://routerunlock.com/download-firmware-huawei-e5172as-22-v200r001c00sp062-free-universal/
3>unpack it it will be show 1 firmware file 
4>flash with balong_flash utility ......
e.g balong_flash -p1 -g5 firmware.BIN and enter it . your device will be flash with forth32 utility.(balong_flash).but recommended to use linux os .because forth32  hate windows and all windows-related stuff.

anix22, 2017-04-16:
because i not use linux and i not know how to use linux
sir i use mehod on windows
it give error

[image]

anix22, 2017-04-16:
[image]

anix22, 2017-04-17:
sir firmware write success but now router power led show pink

and router  wifi not work

when i login the router it show the device as eqipment mode

chipoza, 2017-04-17:
ok this is fw prob.
you find valid fw from googling and try with balong_flash. it will be work..

anix22, 2017-04-17:
sir i check all fw same issue power led stuck on pink led

chipoza, 2017-04-19:
ok dont worry .load usbloader.bin  .2 com port appear . run go_hdlc and then open huawei flasher v2.0 . put comport number and simple flash it . but you need to extrect firmware with this huawei flasher v2.0. (firmware will be in .exe formate)

localhost127001, 2017-04-23:
@chipoza Thank you friend. Your help very much appreciated
I wish you luck with your future endeavors

Mohsan123, 2017-04-24:
@anix22  for extracting firmware use belongflash "belong_flash.exe -e firmware.bin".
I recommended to use belongflash to write firmware with this command. "belongflash -r -g0 firmware.bin"

if you want's to extract firmware with Huawei flasher then use Huawei_flashe v1.6. firmware in may exe or bin format. if your firmware in bin format just change extension .bin to .exe then load in Huawei_flasher v1.6, this will extract it.
This is my poor knowledge that I obtain from internet and mostly studding Sir @forth32 ideas , may be it resolve your problem.

anix22, 2017-04-27:
thanks mohson bro but not work

e5172 diffrent outer it not accept firmware easily and not my issue resolved

bro can you tell me
how to backup nv.image default backup

anix22, 2017-04-30:
ZONG E8372H-153 RESET SOLUTION SPECHIAL THANKS TO VICKY AND nadeem danish

https://drive.google.com/file/d/0B3pEUGa_GFBrcDdxU1FFc2RoRGs/view

closed 2017-09-24 by forth32