forth32/balong-usbdload #28: zong e5573cs-322 not reset please help forth32 sir

anix22, 2017-04-17:
sir i have zong e5573cs-322

after unlock reset button not work

please help me great forth32 sir

chipoza, 2017-04-19:
forth32 not want to show this reset button fixsation solution due to mkey marketing .do not ask next time in this type of question if you want to really fix this you learn more and more ..

anix22, 2017-04-19:
I AM NOT A PROGRAMER THEN HOW YOU SAY LEARN MORE AND MORE

saqi755, 2017-05-06:
Respectable sir i try to unlock E8231s-2 after using original usbloader i succeeded in unlocking but now my device cant show signal of any network no signals issue how can i repair it please someone help me

ronix889, 2017-05-07:
repair imei then check

saqi755, 2017-05-08:
Can any expert  guide me to make a backup of huawei E8231s wingles nv data and restore procedure ? Waiting for a good response from some good people.

Mohsan123, 2017-05-08:
@saqi755  you have to contact nearest Huawei repair center. I have also done this kind of job and lost NV-data unfortunately. After this I contact to my nearest Huawei center for repairing my device.

saqi755, 2017-05-17:
Huawei Center replied in negative

Mohsan123, 2017-05-17:
contact with Huawei on web chat. they will guide best repairing center.

mohish1, 2017-06-17:
https://drive.google.com/file/d/0B7JLyYkSo1rFdmthWmtmMWY1RDg/view?usp=sharing

telenor latest e8372h-608 21.316.01.04.274+reset solution thans to my frend ibrahim who share me reset solution

closed 2017-06-17 by forth32