forth32/balong-usbdload #30: balongnvtool.exe how to use

anix22, 2017-04-19:


renoos, 2017-05-02:
@anix22 
sir i have huawei e5330bs-2 want to unlock . i done boot shot then i run usbdload with cmd command 
[image]

ronix889, 2017-05-06:
anyone help how to use balong nv tool.exe

chipoza, 2017-05-06:
if pc ui interface is com1
balong_usbdload -p1 usblsalf-5330.bin

renoos, 2017-05-07:
@chipoza  after boot shot pin . i have com 12  i try this command balong_usbdload -p12 usblsalf-5330.bin but is not working then i try this command balong_usbdload -c usblsalf-5330.bin i got this error as i was attach the picx . please help me now what i do ?

ronix889, 2017-05-07:
no one have knowledge how to backup nv.imh nv.default nv.dload
strange thing

ronix889, 2017-05-07:
dear renoos bro here no one help to our  normal users seems

ronix889, 2017-05-07:
here only help people which are spechial peoples 

we normal users no one reply no one help

closed 2017-09-24 by forth32