forth32/balong-usbdload #31: how to patch a loader please tell forth32 sir

anix22, 2017-04-28:
forth32 sir please tell
how to patch a loader
i install a ubuntu linux in my pc
and knot know how to patch a loader
please help sir
anyone please help

ronix889, 2017-05-07:
bro here no one help
only help special people
who give money
if anyone comment then truth banned 

then why says it s a punlic form called
money form
i know after my comment forth32 block me
but reality it this
forth32 chipoza help only own frinds which pay a money

ronix889, 2017-05-08:
Why no one reply here in forum 

Strange thing

ronix889, 2017-05-08:
Loader patch source not detail 
How to use 
Then strange thing no reply 
You are a super programmer everyone know 
Normal user are normal 
But no reply 
No reply

chipoza, 2017-05-16:
loader patch source work in linuux .new released usbdloader.bin is already pach one fly...

ValdikSS, 2017-09-23:
This repository contains "loader-patch" automatic patcher to convert usbloader.bin to usblsafe.bin. Moreover, balong-usbdload would patch "unsafe" usb loaders automatically, and if it failed to do so, won't allow you to load unpatched loaders without `-c` flag to prevent flash erasure.

closed 2017-09-24 by forth32