forth32/balong-usbdload #33: HOW TO BACKUP A NV.IMG NV.DEFAULT.NV.DLOAD  IF NO ONE ANSWER IT MEAN WE NORMAL USER NO ONE REPLY

ronix889, 2017-05-07:
HO FORTH32 PLEASE REPLY

HOW TO BACKUP 
NV.IMG NV.DEFAULT NV.DLOAD

IF NOT ANYONE REPLY THEN I THINK 

NO ONE HELP WE LIKE NORMAL PEOPLES

Mohsan123, 2017-05-08:
@ronix889  which device you wants to backup? why not you use safeloader provided by Respectable Sir @forth32 .

ronix889, 2017-05-08:
Dear I use forth 32 sir usblsaf 
But I want know backup 
Some device required nv erased 
So I want backup before use usbloader 
Please help me 
I am very helpless

ronix889, 2017-05-08:
2nd thing is bad 
Why foryh32 sir not reply normal users???
Always ignore public request
We are normal people not a programmer 
If we ask then no one reply ???

Mohsan123, 2017-05-08:
None of devices need erased NV-data when u use Safe-loader Provided by Respectable Sir Forth32.

Mohsan123, 2017-05-08:
i think i know you Mr. @ronix889. Your body language tell me i see you in this form with another name.

ronix889, 2017-05-08:
Mr mohson 
1st thing 
Usbloader not work on stc e5577s-932
Can forth32 patch it 
And where is e5577s-932 patched loader

ronix889, 2017-05-08:
Where is stc e5577s-932 usblsaf loader 
Then you teach me

ronix889, 2017-05-08:
No reply simple thing says that forth 32 can't patch e5577s-932 usblsaf

ronix889, 2017-05-08:
Give my questik. Answer where is e5577s-932 usblsaf.bin 
No usblsaf.bin ????????????
And your behaviour mohson already show
Forth32 only help own peoples 
No care about normal people 
This is my last message in forum 
Because I listen my friends forth 32 is great 
Truthiness is 
Forth32 only help own peoples which pay money 
Bye bye

Mohsan123, 2017-05-08:
Mr. @ronix889 remember we need knowledge. If respectable Sir forth32 give us any idea this their favor on us. if they not reply us, we don't have right to blame them.

ronix889, 2017-05-08:
1st show here e5577s-932 usblsaf.bin 
Then teach me about no need nv erase

ronix889, 2017-05-08:
Mohson we are normal human 
I know forth32 sir is great 
We are normal not a programmer 

Then how we make a usblsaf???
Here I am not a fight with anyone 
Genuine thing I told 
If you publish anything 
Proper detail give it 
You are owner you know everything 
But we are normal users not know programming

Mohsan123, 2017-05-08:
@ronix889 Don't use immortal language. Ask forgiveness to Respectable Sir @forth32 forth32 for your "immortal language".

ronix889, 2017-05-08:
I simple ask help where is e5577s-932 patched loader 
Here no one help

Mohsan123, 2017-05-08:
you can request to Respectable sir for patch this loader.
If they patch loader, this is favor on you if they do not patch, you have no right to blame on them. Respectable sir not complatly free for us, as they have free time, they help everyone on this form.

ronix889, 2017-05-08:
Forth32 sir not reply me

ronix889, 2017-05-08:
I contact on 4pda 
But he not reply my message

ronix889, 2017-05-08:
I give respect forth32 I know he is great 
But why sir not reply we normal persons 
Forth32 sir is no1 in huawei everyone know 
So my request to sir please reply your users on 4pda and even a balong form 
Sorry for my rude behaviour

Mohsan123, 2017-05-08:
Perhaps they busy in their daily life work.
I already tell you that none of us have right to blame on them that they not reply us.
it is their favor on us when they give ideas us.
Now no more comments. And don't creat it as issue.

closed 2017-05-12 by ronix889