forth32/balong-usbdload #34: Dumping original bootloader / firmware? (E5573)

CreateRemoteThread, 2017-05-08:
Hello,

I am trying to follow the work here, to dump the original firmware off a E5573 device. Going through the source code, I can't find a reference to actually obtaining the original bootloader (or dumping anything out).

Is this doable through the balong tools here / balong v7, or is there another way?

Thankyou!

ronix889, 2017-05-08:
HERE NO ONE HELP YOUR BRO
HERE ONLY HELP WHICH PEOPLE PAY A MONEY
JUST NAME PEOPLE FORM ONLY

I ASKING QUESTIONS  NO ONE REPLY HERE

Mohsan123, 2017-05-08:
@CreateRemoteThread  what is exact model of your device?
e5573cs-xxx, e5573s-xxx, e5573bs-xxx

CreateRemoteThread, 2017-05-08:
Hello, the exact model of device I am looking at is an E5573s-606.

(I've found the UART out - and "cshell disable.d.." etc in the boot log, and there's another set of 10 debug pins up top which I'm thinking might be JTAG? Is this the right route to go down?)

chipoza, 2017-05-16:
for dump device firmware run your device in fastboot mode (usb_dlong -p1 -f usblsalf.bin). after 2 com port will be appear
1(linux os)-You should see a single device with PID 36DD. 

2. Manual "pull" fastboot this device one of the already existing in the system serial port drivers, for example, "FC - Pc UI Interface". 

3. Download utility fbrflash here and run it:
fbrflash.exe -p <port_fastboot> -o
It should start the process dump partition flash drive, which is formed as a result of 20 image files.

ronix889, 2017-05-16:
Why make fool to people chippza this method not make any port

ronix889, 2017-05-16:
Manual "pull" fastboot this device one of the already existing in the system serial port drivers, for example, "FC - Pc UI Interface".

ronix889, 2017-05-16:
Download utility fbrflash here and run it:
fbrflash.exe -p <port_fastboot> -o
It should start the process dump partition flash drive, which is formed as a result of 20 image files

Make only 
O file 1 atleast you guide to user correct not misguide

ronix889, 2017-05-16:
I asking how to use nv tool .exe but no one reply ???

ronix889, 2017-05-16:
Forth 32 and chippza are mental patient 
Because 1st they blame mkey 
Then another peoples now 1 person balwinder27
And now blame me I am balwinder27

I think these people braking work stop
Need go to mental hospital and need proper treatment 

Forth32 and chippza are Pakistani Muslims fuckers 
I don't care

closed 2017-09-24 by forth32