forth32/balong-usbdload #44: Please create a feature in dload it automatically  take backup of nv data and restore after loader flash

gmbest89, 2017-09-29:
@forth32  @rust3028  @ValdikSS

ValdikSS, 2017-09-29:
I assume you're talking about stock unpatched usbloaders.
This is not trivial. there isn't any standardized way to dump nvram partitions content prior to loading usbloader. Modified usbloader can dump and restore nvram partitions, but if you're limited only to unpacked loaders due to secboot, you can't load them.

You can always re-sign your current modified firmware the way to enable i.e. telnetd, with leaked Huawei signature keys, flash it to your device, dump nvram partitions and flash any unsafe usbloaders you want. But still, with secboot enabled you don't really have many choice.

gmbest89, 2017-09-29:
first all of thanks @ValdikSS for Quick response i have done modifications in firmware but i did not found leaked signature keys  please provide me IN PM gmbest89@gmail.com